ch32-rs/ch32-hal #3: Is it possible to build it with only crates.io version of deps?

alexxy, 2024-03-19:
Hi!

Is it possible to build it with only crates.io version of deps? Or are there some instructions how to get all needed deps?

andelf, 2024-03-20:
Sorry, the docs aren't complete yet. This is still a work-in-progress, so some peripherals might have issues.

```bash
rustup default nightly

rustup target add riscv32imac-unknown-none-elf
rustup target add riscv32imafc-unknown-none-elf

cd ../

git clone https://github.com/ch32-rs/ch32-data.git
git clone https://github.com/ch32-rs/chiptool.git

(cd ch32-data && ./d gen)
# generates the metapac.
```

andelf, 2024-03-20:
I've also added the metapac repo: "https://github.com/ch32-rs/ch32-metapac". It's built automatically from the newest branch of ch32-data.

andelf, 2024-04-30:
Close as fixed

closed 2024-04-30 by andelf

andelf, 2024-05-11:
Now this crate introduces ch32-metapac via git tag.