ch32-rs/ch32-hal #6: V307: not running

romainreignier, 2024-04-08:
I have tried several examples of the CH32V307 on a "CH32V307 EVT by SCDZ" board:

[image]

But the programs are not running (not blinks with blinky.rs and no output with `sdi_print.rs`).
I have tried to change the HSI/HSE settings while initializing the hal but nothing changed.
I the V307 supposed to work already? Or maybe a regression introduced with recent changes?
I have pulled the ch32-metapac from the [prebuilt repo](https://github.com/ch32-rs/ch32-metapac).
I have tried the [blink.c example](https://github.com/cnlohr/ch32v003fun/tree/master/examples_v30x/blink) from ch32v003fun and it works.

It might be interesting to enable the "Discussions" on this repo to avoid opening issues for these kind of newbie questions.

Thanks

andelf, 2024-04-10:
> no output with sdi_print.rs

Then the whole dev env setting doesn't work.

You should remove "SDI Print" and any "print!()" in the demo code.
Since the WCH-Link you provide doesn't support SDI Print. And SDI print's mem waiting will hang forever.

andelf, 2024-04-10:
> It might be interesting to enable the "Discussions"

romainreignier, 2024-04-10:
> You should remove "SDI Print" and any "print!()" in the demo code.
> Since the WCH-Link you provide doesn't support SDI Print. And SDI print's mem waiting will hang forever.

I use the WCH-LinkE from my CH32V003 dev board with the v307. The wlink properly warn about the missing SDI feature of the default probe :)

I will continue to investigate.

andelf, 2024-04-11:
`wlink regs` might be helpful for debugging if no SDI print is available.
If SDI print is OK, you can add `println!` to `panic_handle`.

romainreignier, 2024-04-13:
I have done some tests today and I have noticed that the config that works in the `ch32v003fun` project uses a configuration for the CH32V30**5RBT6** and not the CH32V30**7VCT6** present on my board.
But switching to the CH32V30**7VCT6** in the Makefile prevents the code to run.
See https://github.com/cnlohr/ch32v003fun/issues/303
I have then switched `ch32-hal` feature to `ch32v305rbt6` and the examples started to run (blinky, uart, button, exti)!
I don't understand the issue because from what I understand, only the size of the FLASH and RAM in the linker script is changed between these two configs (appart for the missing peripherals, not used in simple examples).

I have noticed that with my WCH-Link-CH549 probe, the code run even with `hal::debug::SDIPrint::enable();` and `println!()` present in code.

romainreignier, 2024-06-17:
I have received a new CH32V307VCT6 board (a YD-CH32V307VCT6 clone) and I am facing the same issue, no code run on it.
Using `ch32v305rbt6` feature, I have managed to get some non-embassy code to run, otherwise the program seems to block quickly (I can see some sdi logs at the start of the main).

romainreignier, 2024-06-17:
`wlink regs`:
```
13:21:50 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
13:21:50 [INFO] Attached chip: CH32V30X [CH32V307VCT6] (ChipID: 0x30700528)
13:21:50 [INFO] Dump GPRs
dpc(pc):   0x00000260
x0   zero: 0x00000000
x1     ra: 0x000001d8
x2     sp: 0x20010000
x3     gp: 0x20000800
x4     tp: 0xb2e10030
x5     t0: 0x00000000
x6     t1: 0x00200000
x7     t2: 0x00002000
x8     s0: 0x0000c351
x9     s1: 0xe000e048
x10    a0: 0x2000024c
x11    a1: 0x200002d4
x12    a2: 0x20000020
x13    a3: 0xfffeffff
x14    a4: 0x00000000
x15    a5: 0x40022000
x16    a6: 0x00010000
x17    a7: 0x00000040
x18    s2: 0x000013a0
x19    s3: 0x00000001
x20    s4: 0x00008000
x21    s5: 0x000012c4
x22    s6: 0x5034ec5a
x23    s7: 0x4f2c228d
x24    s8: 0x68ca152b
x25    s9: 0xac10a080
x26   s10: 0x525e0889
x27   s11: 0xcb5d3c60
x28    t3: 0x20001700
x29    t4: 0xd0006425
x30    t5: 0x89a82635
x31    t6: 0x14fca411
marchid  : 0xdc68d881
mimpid   : 0xdc688001
mhartid  : 0x00000000
misa     : 0x40901125
mtvec    : 0x00000000
mscratch : 0x020c4406
mepc     : 0x000001d2
mcause   : 0x00000007
mtval    : 0x2000ffdc
mstatus  : 0x00000000
dcsr     : 0x400000c3
dpc      : 0x00000260
dscratch0: 0x0000c351
dscratch1: 0x2000024c
gintenr  : 0x00000000
intsyscr : 0x00000000
corecfgr : 0x00000000
```

`wlink status`:
```
13:22:05 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
13:22:05 [INFO] Attached chip: CH32V30X [CH32V307VCT6] (ChipID: 0x30700528)
13:22:05 [INFO] Chip ESIG: FlashSize(288KB) UID(a3-65-be-7b-54-50-b5-31)
13:22:05 [INFO] Flash protected: false
13:22:05 [INFO] RISC-V ISA(misa): Some("RV32ACFIMUX")
13:22:05 [INFO] RISC-V arch(marchid): Some("WCH-V4A")
13:22:05 [WARN] The halt status may be incorrect because detaching might resume the MCU
13:22:05 [INFO] Dmstatus {
    .0: 0x382,
    allhavereset: false,
    anyhavereset: false,
    allresumeack: false,
    anyresumeack: false,
    allunavail: false,
    anyunavail: false,
    allrunning: false,
    anyrunning: false,
    allhalted: true,
    anyhalted: true,
    authenticated: true,
    version: 0x2,
}
13:22:05 [INFO] Dmcontrol {
    .0: 0x80000001,
    haltreq: true,
    resumereq: false,
    ackhavereset: false,
    ndmreset: false,
    dmactive: true,
}
13:22:05 [INFO] Hartinfo {
    .0: 0x312380,
    nscratch: 0x3,
    dataaccess: true,
    datasize: 0x2,
    dataaddr: 0x380,
}
13:22:05 [INFO] Abstractcs {
    .0: 0x8000002,
    progbufsize: 0x8,
    busy: false,
    cmderr: 0x0,
    datacount: 0x2,
}
13:22:05 [INFO] haltsum0: 0x1
```

andelf, 2024-06-17:
> mcause   : 0x00000007
> mtval    : 0x2000ffdc

It's a StoreFault error when accessing 0x2000ffdc.

Could you help diagnose which instruction causes the error?

Hint:

`mepc     : 0x000001d2` and `cargo objdump` helps.

romainreignier, 2024-06-17:
I can help but I don't understand what are you asking for exactly? I am not too confident in low level embedded stuff.

romainreignier, 2024-06-17:
I have tried:
```
cargo +nightly objdump --release --bin=blinky  --  -S --start-address=0x000001d2
```
That gave me:
```
Disassembly of section .vector_table:

000001d0 <_exception_handler>:
     1d2: 06 c0         sw      ra, 0x0(sp)
     1d4: ef 00 c0 00   jal     0x1e0 <.Lline_table_start0+0x29>
     1d8: 82 40         lw      ra, 0x0(sp)
     1da: 11 01         addi    sp, sp, 0x4
     1dc: 73 00 20 30   mret
[...]
```

andelf, 2024-06-18:
I'll run some stress tests on the blinky and the board to investigate further. 
Debugging can be tricky, but I'll keep this thread updated on my progress.

romainreignier, 2024-06-18:
From what I have seen, no code run when the FLASH size is set to 256k and RAM size to 64k.
The CH32V305RBT6 is quite close to the CH32V307VCT6, some peripherals are missing (CAN, Ethernet, some UART) and the memory is 128k / 32k.
According to the datasheet (mentioned here: https://github.com/cnlohr/ch32v003fun/issues/303#issuecomment-2074884014), the memory can be split differently. I have then tried the 192k/128k with no success but the 288k/32k produces the same effect than using the `ch32v305rbt6` feature.

This code runs correctly:

```rust
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use hal::gpio::{Level, Output};
use {ch32_hal as hal, panic_halt as _};
use qingke::riscv;

#[qingke_rt::entry]
fn main() -> ! {
    let p = hal::init(Default::default());
    let mut led = Output::new(p.PA15, Level::Low, Default::default());
    loop {
        led.toggle();
        riscv::asm::delay(1000000);
    }
}
```

But the embassy blinky (from the examples) blocks after having switched ON the LEDs.

`wlink regs`:

```
07:50:08 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
07:50:08 [INFO] Attached chip: CH32V30X [CH32V307VCT6] (ChipID: 0x30700528)
07:50:08 [INFO] Dump GPRs
dpc(pc):   0x00000880
x0   zero: 0x00000000
x1     ra: 0x00000860
x2     sp: 0x20007fa0
x3     gp: 0x20000800
x4     tp: 0xb2e10020
x5     t0: 0x000000c5
x6     t1: 0x00200000
x7     t2: 0x00002000
x8     s0: 0x20000020
x9     s1: 0x00000001
x10    a0: 0x00002088
x11    a1: 0x00000000
x12    a2: 0x20000000
x13    a3: 0x00000000
x14    a4: 0x00018737
x15    a5: 0xe000f000
x16    a6: 0x00000000
x17    a7: 0x00000000
x18    s2: 0x000010b0
x19    s3: 0x20007ff4
x20    s4: 0x00000000
x21    s5: 0x20000110
x22    s6: 0x20007fec
x23    s7: 0xffffffff
x24    s8: 0xfffffffb
x25    s9: 0xfffffffd
x26   s10: 0x00000004
x27   s11: 0x00000000
x28    t3: 0x20001700
x29    t4: 0xd0006425
x30    t5: 0x89a82635
x31    t6: 0x04f8a091
marchid  : 0xdc68d881
mimpid   : 0xdc688001
mhartid  : 0x00000000
misa     : 0x40901125
mtvec    : 0x00000003
mscratch : 0x020c6406
mepc     : 0x00000b34
mcause   : 0x8000000c
mtval    : 0x00000000
mstatus  : 0x00002088
dcsr     : 0x400000c0
dpc      : 0x00000880
dscratch0: 0x20000020
dscratch1: 0x00002088
gintenr  : 0x00002088
intsyscr : 0x00000003
corecfgr : 0x0000001f
```

`wlink status`

```
07:57:15 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
07:57:15 [INFO] Attached chip: CH32V30X [CH32V307VCT6] (ChipID: 0x30700528)
07:57:15 [INFO] Chip ESIG: FlashSize(288KB) UID(a3-65-be-7b-54-50-b5-31)
07:57:15 [INFO] Flash protected: false
07:57:15 [INFO] RISC-V ISA(misa): Some("RV32ACFIMUX")
07:57:15 [INFO] RISC-V arch(marchid): Some("WCH-V4A")
07:57:15 [WARN] The halt status may be incorrect because detaching might resume the MCU
07:57:15 [INFO] Dmstatus {
    .0: 0x382,
    allhavereset: false,
    anyhavereset: false,
    allresumeack: false,
    anyresumeack: false,
    allunavail: false,
    anyunavail: false,
    allrunning: false,
    anyrunning: false,
    allhalted: true,
    anyhalted: true,
    authenticated: true,
    version: 0x2,
}
07:57:15 [INFO] Dmcontrol {
    .0: 0x80000001,
    haltreq: true,
    resumereq: false,
    ackhavereset: false,
    ndmreset: false,
    dmactive: true,
}
07:57:15 [INFO] Hartinfo {
    .0: 0x312380,
    nscratch: 0x3,
    dataaccess: true,
    datasize: 0x2,
    dataaddr: 0x380,
}
07:57:15 [INFO] Abstractcs {
    .0: 0x8000002,
    progbufsize: 0x8,
    busy: false,
    cmderr: 0x0,
    datacount: 0x2,
}
07:57:15 [INFO] haltsum0: 0x1
```

I see that the cause is different: `mcause   : 0x8000000c`

And `mepc     : 0x00000b34`

```
cargo +nightly objdump --release --bin=blinky  --  -S --start-address=0x00000b34
```

```
00000abc <_embassy_time_set_alarm>:
     b34: 73 26 00 80   csrr    a2, 0x800
     b38: 21 89         andi    a0, a0, 0x8
     b3a: 51 8d         or      a0, a0, a2
     b3c: 73 20 05 80   csrs    0x800, a0
     b40: 2e 85         mv      a0, a1
     b42: b2 40         lw      ra, 0xc(sp)
     b44: 22 44         lw      s0, 0x8(sp)
     b46: 92 44         lw      s1, 0x4(sp)
     b48: 41 01         addi    sp, sp, 0x10
     b4a: 82 80         ret
```

romainreignier, 2024-06-18:
After a git bisect, it appears that the failing embassy examples on ch32v307 are caused by the change of the embassy_time feature from `tick-hz-10_000` to `tick-hz-1_000_000`.
I have tested `tick-hz-100_000` and the code stops working after about 3 seconds with the `wlink regs`:

```
09:33:55 [INFO] Connected to WCH-Link v2.12(v32) (WCH-LinkE-CH32V305)
09:33:55 [INFO] Attached chip: CH32V30X [CH32V307VCT6] (ChipID: 0x30700528)
09:33:55 [INFO] Dump GPRs
dpc(pc):   0x000008d2
x0   zero: 0x00000000
x1     ra: 0x000008b2
x2     sp: 0x20007fa0
x3     gp: 0x20000800
x4     tp: 0xb2e10020
x5     t0: 0x0004eb49
x6     t1: 0x0000157a
x7     t2: 0x00002000
x8     s0: 0x20000068
x9     s1: 0x00000001
x10    a0: 0x00002088
x11    a1: 0x00000000
x12    a2: 0x20000000
x13    a3: 0x00000000
x14    a4: 0x0032b720
x15    a5: 0xe000f000
x16    a6: 0x00000000
x17    a7: 0x00000000
x18    s2: 0x000013de
x19    s3: 0x20007ff4
x20    s4: 0x00000000
x21    s5: 0x20000038
x22    s6: 0x20007fec
x23    s7: 0xffffffff
x24    s8: 0xfffffffb
x25    s9: 0xfffffffd
x26   s10: 0x00000004
x27   s11: 0x00000000
x28    t3: 0x20001c00
x29    t4: 0x00000025
x30    t5: 0x00000025
x31    t6: 0x04f8a091
marchid  : 0xdc68d881
mimpid   : 0xdc688001
mhartid  : 0x00000000
misa     : 0x40901125
mtvec    : 0x00000003
mscratch : 0x020c6406
mepc     : 0x00000812
mcause   : 0x8000000c
mtval    : 0x00000000
mstatus  : 0x00002088
dcsr     : 0x400000c0
dpc      : 0x000008d2
dscratch0: 0x20000068
dscratch1: 0x00002088
gintenr  : 0x00002088
intsyscr : 0x00000003
corecfgr : 0x0000001f
```

romainreignier, 2024-06-18:
By looking at systick I have seen that with the default hal init `let p = hal::init(Default::default());` the HSI of 8 MHz is used without PLL and the embassy time driver systick is using the HCLK/8 configuration of the systick so the timer runs at 1 MHz which is the same speed as configured for the embassy_time with the feature `tick-hz-1_000_000`.
Using another configuration like `hal::rcc::Config::SYSCLK_FREQ_96MHZ_HSI` allows to run with `tick-hz-1_000_000`.

So either the embassy_time tick rate should be reduced or all the examples updated to use a PLL of Systick timer set in mode HCLK without the div 8.

romainreignier, 2024-06-18:
> the examples updated to use a PLL

The blinky blocked after a while (`SYSCLK_FREQ_96MHZ_HSI` + `tick-hz-1_000_000`)

> Systick timer set in mode HCLK without the div 8

Does not seems to work

romainreignier, 2024-07-12:
I think that I have finally found an answer to my issue with the FLASH and RAM size in the linker script.
As mentioned before, the size of the FLASH and RAM can be selectable but I did not known how.
Thanks to Martin and TheSnide on  cnlohr Discord, I have now understood that the size is selected by some USER config bits that can be queried with wchisp:
```sh
$ wchisp config info
11:10:00 [INFO] Current config registers: a55aff0000ff00ffffffffff00020900a365be7b5450b531
RDPR_USER: 0x00FF5AA5
  [7:0]   RDPR 0xA5 (0b10100101)
    `- Unprotected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x3 (0b11)
    `- CODE-228KB + RAM-32KB
DATA: 0xFF00FF00
  [7:0]   DATA0 0x0 (0b0)
  [23:16] DATA1 0x0 (0b0)
WRP: 0xFFFFFFFF
  `- Unprotected
```

On my board, the config is `CODE-228KB + RAM-32KB` which matches what I have found with trial and errors https://github.com/ch32-rs/ch32-hal/issues/6#issuecomment-2175453942

Now I need to figure out how to change that setting and add some info in the CH32V307 example README about it.

romainreignier, 2024-07-12:
I have read the current value this code and it works:
```rust
let flash = hal::pac::FLASH;
let sram_code_mode = flash.obr().read().sram_code_mode();
hal::println!("sram_code_mode = {:#b}", sram_code_mode);
```

But I have failed to set a new value with this code:
```rust
flash.obr().modify(|w| w.set_sram_code_mode(0b10));
```
I have a build error:
```rust
error[E0599]: the method `modify` exists for struct `Reg<Obr, R>`, but its trait bounds were not satisfied
  --> src/main.rs:25:17
   |
25 |     flash.obr().modify(|w| w.set_sram_code_mode(0b10));
   |                 ^^^^^^ method cannot be called on `Reg<Obr, R>` due to unsatisfied trait bounds
   |
  ::: /home/rre/.cargo/git/checkouts/ch32-metapac-06b85fb47469f0da/82e1fdc/src/common.rs:6:1
   |
6  | pub struct R;
   | ------------ doesn't satisfy `R: ch32_hal::ch32_metapac::common::Write`
   |
   = note: the following trait bounds were not satisfied:
           `R: ch32_hal::ch32_metapac::common::Write`
```

I am not used to use direct pac access, I must have overlooked something but I don't know what.

andelf, 2024-07-12:
@romainreignier FLASH_OBR is access-only according to datasheet.

andelf, 2024-07-12:
It requires unlocking and programming sequence using FLASH_CTLR.

romainreignier, 2024-07-12:
Ok, thanks, I will have to write the flash sequence then.

andelf, 2024-11-09:
Close as fixed. Thank you all.

closed 2024-11-09 by andelf