ch32-rs/ch32-hal #11: Is there an approach to connect ethernet (wchnet)?

christopherkutzmann, 2024-04-25:
Hello!
This project is awesome and must be driven forward. I have successfully port my interrupt switch logic from EmbeetleIDE (c) to this format and rust. It is much cleaner and it consumes less current. Now i am facing the final task: to send short mqtt messages from the ch32v307 (CH32V307V-R1-1v0). I already have it working in an Embeetle project in c, but now I want to use ethernet with rust and this embassy. I've already read a lot about embassy-net(-driver/-channel etc.), but haven't found the right solution for me.
I think I can also just include the wch-ethernet library and use it, but I'm really looking for the right way to do it.
Does anyone have an idea or even a solution?

Board: [https://github.com/openwch/ch32v307/tree/main/SCHPCB/CH32V307V-R1-1v0](https://github.com/openwch/ch32v307/tree/main/SCHPCB/CH32V307V-R1-1v0)
Ethernet example: [https://github.com/openwch/ch32v307/tree/main/EVT/EXAM/ETH/MQTT](https://github.com/openwch/ch32v307/tree/main/EVT/EXAM/ETH/MQTT)
Ethernet Library: [https://github.com/openwch/ch32v307/tree/main/EVT/EXAM/ETH/NetLib](https://github.com/openwch/ch32v307/tree/main/EVT/EXAM/ETH/NetLib)

Many thanks,
Chris

andelf, 2024-04-26:
Thanks. Ethernet is not implemented yet.
And the EVT seems to have some close source code(the Net lib).
I'll check this after the DMA part is implemented.

christopherkutzmann, 2024-06-12:
Dear @andelf,
I've just looked in again after a few weeks and see that the DMA is making progress.
What is the status of the ethernet example? Please inform us!
Regards, Chris

chmousset, 2025-01-03:
@andelf has there been any significant advancement on the Ethernet part?

I've been looking a bit into Ethernet for the CH32V208 on my side. As [others pointed out](https://www.eevblog.com/forum/microcontrollers/ch32v307-risc-v-minicore-with-ethernet/msg5181855/#msg5181855), it seems that STM32F4xx code should work on the CH32.

So I was planning on frankensteining the [STM32](https://github.com/embassy-rs/embassy/tree/main/embassy-stm32/src/eth) with ch32-hal, but if you already have a WIP maybe I can help.

christopherkutzmann, 2025-01-25:
I am also still in the process of finding a solution. I have used the time to implement and connect further modules. Now all that's really missing is ethernet!

@chmousset Are there any initial results with the Frankenstein Monster?

chmousset, 2025-01-25:
@christopherkutzmann not much. I did a little digging and here are my notes:

CH32 have two Ethernet controllers:
- A 10M MAC + PHY, no MII interface - present in the CH32V208, CH579 and a few other uC (RM section 27.2)
- A 10/100/1G MAC + 10M PHY or 10/100M PHY - present in the other lines of uC (RM section 27.1)

## 10M MAC (CH32V208, CH579)
This one is simpler than the STM32's. It's barely more complex than an SPI controller, so it can be implemented from scratch relatively quickly.

Not that I found any usable reference anyway...
MournRiverStudio doesn't have HAL for the 32V208 's Ethernet peripheral, and the included SVD doesn't show the same registers as the doc.

It's also a bit restrictive on the system clock, as FSYS has to be 60 or 120MHz.

**status**: I'm in the process of adding the register and peripheral definition to [ch32-data](https://github.com/ch32-rs/ch32-data), and will probably try to get a link up and running tomorrow.

## 10/100/1G MAC (CH32V307)
Comparing STM32 RM and CH32 RM indicate that the this MAC is most likely compatible with `eth_v1a`  driver from embassy-stm32:
- present in v1b and v1c, but not in doc:
  - no mention of regs::Dmarswtr
  - no mention of edfe
- present in v1, but not in doc:
  - no mention of MLT
  - no mention of dmasbmr
  
disclaimer: I didn't check all the registers

**status**: Since I don't intend on using any uC using this MAC, I stopped working on that.

chmousset, 2025-01-26:
TX works on the CH32V208!
Well, sort of: there is still a lot of work to be done before there is a clean HAL though.

experimentation project: https://github.com/chmousset/rs-ch32v208-eth
MAC-10M register addition: https://github.com/ch32-rs/ch32-data/pull/16

[image]

jaxter184, 2025-04-04:
I was hoping I'd be able to integrate @chmousset 's changes with `embassy-net-driver-channel` by copy/pasting code from various examples, but it's become clear that I have no idea what I'm doing. I've got my current progress here: https://github.com/chmousset/rs-ch32v208-eth/compare/master...jaxter184:rs-ch32v208-eth:main and right now it seems to stall when trying to spawn tasks.

It looks like the API of `embassy-net` has changed a bit between v0.5 and v0.6, and v0.6 requires `embassy-time` v0.4, so maybe that should get bumped first?

Thanks for all y'all's efforts so far!

chmousset, 2025-04-04:
@jaxter184 I forgot to mention it here: I got it working. I have an example in PR #89 
Indeed `embassy-time` and `embassy-net` ideally needs a bump, but I didn't have time to dig into these crates

jaxter184, 2025-04-06:
@chmousset Awesome! It seems to run, but I'm not able to get past the DHCP step. The print message claims to have sent the packet, but I'm not able to see anything on wireshark, and the port light on my switch isn't turning on (with the ARP example, I was able to get both). I'll do a little more poking around to see if the issue is something on my end. Either way, this looks great, thanks for following up!