ch32-rs/ch32-hal #12: V003:Stuck at initializing UartTx with PC0 pin

Dicklessgreat, 2024-04-29:
I'm using `CH32V003F4P6`.
```
let p = ch32_hal::init(Default::default());
println!("initialize uart tx");
let mut uart = UartTx::new(p.USART1, p.PC0, Default::default()).unwrap();
println!("finished initializing uart");
```
This prints only `initialize uart`.
Note that it's not due to panic caused by `unwrap()` that `finished initializing uart` is not printed.
I tested code and found it stuck [at this line in UartTx::new_inner()](https://github.com/ch32-rs/ch32-hal/blob/4ed71ae078dc1661ad3ce9d430394f743d37a5f5/src/usart/mod.rs#L124). 

And I also found `PD5` works fine.
```
let p = ch32_hal::init(Default::default());
println!("initialize uart tx");
let mut uart = UartTx::new(p.USART1, p.PD5, Default::default()).unwrap();
println!("finished initializing uart");
```
This code prints.
```
initialize uart tx
finished initializing uart
```
And it works fine for me now.

`CH32V003F4P6` pinouts diagram from datasheet p.10 says `PC0` can be `UTX_`(I don't know what this `_` means, though), so I think it can be used for `UartTx`(and it can compile with PC0, so you expect it to be used?).
Let me correct if I'm wrong:)

andelf, 2024-04-29:
> I don't know what this _ means, though
Underline means "available as REMAP function", then you should check "AFIO" part in the RM pdf.

I'm on vacation without a CH32V003 dev board at hand. I'll check this later.

A few hints:

- use `wlink dump 0x40013800 0x10` to debug the registers.
- use `wlink regs` to dump current status (PC, mstatus are helpful)
- use `cargo objdump --release --bin demo -- -d` to dump the assembly code, the "PC" aligns with address here
- you can also add as many `crate::println!` in the hal crate to debug.

Dicklessgreat, 2024-04-30:
Thanks for your great hints!

Hmm...as far as I can tell, everything do as what it should be but it stucks at calling `set_remap()`.

Trying to initializing `UartTx` with `PC0`, I added `println!("{}", REMAP);` into `UartTx::new_inner()`, then it prints 3. 
3 completely matches what datasheet says to remap PC0 as UART.

After it got stucked, I did `$ wlink dump 0x40013800 0x10`.
```
11:14:19 [INFO] Connected to WCH-Link v2.11(v31) (WCH-LinkE-CH32V305)
11:14:19 [INFO] Attached chip: CH32V003 [CH32V003F4P6] (ChipID: 0x00300500)
11:14:19 [INFO] Read memory from 0x40013800 to 0x40013810
Length: 16 (0x10) bytes | printable whitespace ascii_other non_ascii
40013800:   c0 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00   ×000000000000000
```
And here result of `wlink regs`
```
11:25:54 [INFO] Connected to WCH-Link v2.11(v31) (WCH-LinkE-CH32V305)
11:25:54 [INFO] Attached chip: CH32V003 [CH32V003F4P6] (ChipID: 0x00300500)
11:25:54 [INFO] Dump GPRs
dpc(pc):   0x00000cac
x0   zero: 0x00000000
x1     ra: 0x00000992
x2     sp: 0x200007d4
x3     gp: 0x20000800
x4     tp: 0x1c28040d
x5     t0: 0x00000436
x6     t1: 0x00000cdc
x7     t2: 0x00080000
x8     s0: 0x40021000
x9     s1: 0xffffbfff
x10    a0: 0xe00000f4
x11    a1: 0x40013810
x12    a2: 0x00000000
x13    a3: 0x4001380c
x14    a4: 0x00000001
x15    a5: 0x200007d4
marchid  : 0xdc68d841
mimpid   : 0xdc688001
mhartid  : 0x00000000
misa     : 0x40800014
mtvec    : 0x00000000
mscratch : 0x21410bd1
mepc     : 0x00000436
mcause   : 0x00000007
mtval    : 0x00000000
mstatus  : 0x00001888
dcsr     : 0x400000c3
dpc      : 0x00000cac
dscratch0: 0xe00000f4
dscratch1: 0x40013810
gintenr  : 0x00000000
intsyscr : 0x00000003
corecfgr : 0x00000000
```
So, nothing had been changed.
Maybe datasheet description bug??

andelf, 2024-04-30:
> mcause   : 0x00000007

`macuse[31] = 0`, means it's an exception.

Refer QingkeV2 Processor Manual: exception id=7: it's a StoreFault.

Then you need to diagnose what causes the StoreFault exception.

andelf, 2024-05-05:
This is caused by complex remap rules in USART1.

closed 2024-05-05 by andelf

Dicklessgreat, 2024-05-06:
Confirmed that it works on my chip!!
Thanks.

commit 9405001b (2024-05-05): fix: v0 usart1 remap

Close #12