ch32-rs/ch32-hal #20: Missing instructions for how to build and deploy and example application

ryan-kraay, 2024-05-27:
Hiya,

I've got a CH32V203C8T6 and I'm really interested in exploring what  the Embassy framework has to offer (it looks like an awesome alternative to running a rtos).

However, I'm struggling to get one of the [example](./examples/ch32v203) applications to work.  They compile, just fine, but when I deploy them - they don't seem to do anything (ie: blinky doesn't blink).

I'm not very familiar with Rust (but are I'm using this project as an excuse to explore it).

Here's what I've done so far (most of which was taken from #3 ):

```bash
rustup default nightly
rustup target add riscv32imac-unknown-none-elf

cd examples/ch32v203
cargo build --target riscv32imac-unknown-none-elf --release
# on the ch32v2, I hold the power + reset, then release reset and power
wchisp flash target/riscv32imac-unknown-none-elf/release/blinky
# no blinking on the ch32v2 :(
```

I've also changed [this line](https://github.com/ch32-rs/ch32-hal/blob/f17d8bab1f0161eb200276b33bfc2c39e184ff19/examples/ch32v203/Cargo.toml#L8) to refer to `ch32v203c8t6` and recompiled blinky... and again... nothing blinked.

Am I missing something?  Or could it be that the example code doesn't work with my chip (for whatever reason).

Thanks for the help!  I'm sure it'll help others too :)

andelf, 2024-05-29:
Try the following steps to diagnose the issue:

- Make sure the `Output::new(p.PB8, Level::Low, Default::default())` matches the LED pin of the board
- Try flash and debug using WCH-Link, which has more debugging commands

I've tested the demo code with nanoCH32V203 board. <https://github.com/wuxx/nanoCH32V203/blob/master/hardware/nanoCH32V203.pdf>

Session with `wlink`:

```
wlink -v flash --enable-sdi-print --watch-serial --erase target/riscv32imac-unknown-none-elf/release/blinky
04:22:22 [DEBUG] (1) wlink::usb_device::libusb: Serial number: "2FC58F06B4A9"
04:22:22 [INFO] Connected to WCH-Link v2.11(v31) (WCH-LinkE-CH32V305)
04:22:22 [INFO] Attached chip: CH32V20X [CH32V203C8T6] (ChipID: 0x20310510)
04:22:22 [INFO] Chip ESIG: FlashSize(64KB) UID(cd-ab-51-fc-5b-bc-79-64)
04:22:22 [INFO] Flash protected: false
04:22:22 [DEBUG] (1) wlink::operations: SRAM CODE split mode: 2
04:22:22 [INFO] Erase Flash
04:22:22 [DEBUG] (1) wlink::operations: Reattach chip
04:22:22 [DEBUG] (1) wlink::operations: Reattach chip
04:22:22 [INFO] Read protected: false
04:22:22 [INFO] Read target/riscv32imac-unknown-none-elf/release/blinky as ELF format
04:22:22 [DEBUG] (1) wlink::firmware: Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x5
04:22:22 [DEBUG] (1) wlink::firmware: Matching section: ".vector_table" offset: 0x1000 size: 0x16a
04:22:22 [DEBUG] (1) wlink::firmware: Matching section: ".text" offset: 0x116c size: 0x6d4
04:22:22 [DEBUG] (1) wlink::firmware: Section names: [".vector_table", ".text"]
04:22:22 [DEBUG] (1) wlink::firmware: Found loadable segment, physical address: 0x00000840, virtual address: 0x00000840, flags: 0x4
04:22:22 [DEBUG] (1) wlink::firmware: Matching section: ".rodata" offset: 0x1840 size: 0x9c
04:22:22 [DEBUG] (1) wlink::firmware: Section names: [".rodata"]
04:22:22 [DEBUG] (1) wlink::firmware: Found loadable segment, physical address: 0x000008e0, virtual address: 0x20000000, flags: 0x6
04:22:22 [DEBUG] (1) wlink::firmware: Matching section: ".data" offset: 0x2000 size: 0x8
04:22:22 [DEBUG] (1) wlink::firmware: Section names: [".data"]
04:22:22 [DEBUG] (1) wlink::firmware: found 3 sections
04:22:22 [DEBUG] (1) wlink::firmware: Merge firmware sections with gap: 4
04:22:22 [INFO] Flashing 2280 bytes to 0x08000000
04:22:22 [DEBUG] (1) wlink::operations: Reattach chip
04:22:22 [DEBUG] (1) wlink::operations: Reattach chip
04:22:22 [INFO] Read protected: false
04:22:22 [DEBUG] (1) wlink::operations: Using write pack size 4096 data pack size 256
04:22:22 [DEBUG] (1) wlink::operations: Flash OP written
04:22:22 [DEBUG] (1) wlink::operations: Fastprogram done
04:22:22 [INFO] Flash done
04:22:23 [INFO] Now reset...
04:22:23 [INFO] Now connect to the WCH-Link serial port to read SDI print
04:22:23 [DEBUG] (1) wlink::probe: Opening serial port: "/dev/cu.usbmodem2FC58F06B4A92"
```

Session with `wchisp`:

```
wchisp flash ./target/riscv32imac-unknown-none-elf/release/blinky
04:23:43 [INFO] Chip: CH32V203C8T6[0x3119] (Code Flash: 64KiB)
04:23:43 [INFO] Chip UID: CD-AB-51-FC-5B-BC-79-64
04:23:43 [INFO] BTVER(bootloader ver): 02.60
04:23:43 [INFO] Code Flash protected: false
04:23:43 [INFO] Current config registers: a55abf40ff00ff00ffffffff00020600cdab51fc5bbc7964
RDPR_USER: 0x40BF5AA5
  [7:0]   RDPR 0xA5 (0b10100101)
    `- Unprotected
  [16:16] IWDG_SW 0x1 (0b1)
    `- IWDG enabled by the software, and disabled by hardware
  [17:17] STOP_RST 0x1 (0b1)
    `- Disable
  [18:18] STANDBY_RST 0x1 (0b1)
    `- Disable, entering standby-mode without RST
  [23:22] SRAM_CODE_MODE 0x2 (0b10)
    `- CODE-256KB + RAM-64KB / CODE-160KB + RAM-32KB depending on the chip
DATA: 0x00FF00FF
  [7:0]   DATA0 0xFF (0b11111111)
  [23:16] DATA1 0xFF (0b11111111)
WRP: 0xFFFFFFFF
  `- Unprotected
04:23:43 [INFO] Read ./target/riscv32imac-unknown-none-elf/release/blinky as ELF format
04:23:43 [INFO] Found loadable segment, physical address: 0x00000000, virtual address: 0x00000000, flags: 0x5
04:23:43 [INFO] Section names: [".vector_table", ".text"]
04:23:43 [INFO] Found loadable segment, physical address: 0x00000840, virtual address: 0x00000840, flags: 0x4
04:23:43 [INFO] Section names: [".rodata"]
04:23:43 [INFO] Found loadable segment, physical address: 0x000008e0, virtual address: 0x20000000, flags: 0x6
04:23:43 [INFO] Section names: [".data"]
04:23:43 [INFO] Firmware size: 3072
04:23:43 [INFO] Erasing...
04:23:43 [WARN] erase_code: set min number of erased sectors to 8
04:23:43 [INFO] Erased 8 code flash sectors
04:23:44 [INFO] Erase done
04:23:44 [INFO] Writing to code flash...
3072/307204:23:44 [INFO] Code flash 3072 bytes written
04:23:45 [INFO] Verifying...
3072/307204:23:45 [INFO] Verify OK
04:23:45 [INFO] Now reset device and skip any communication errors
04:23:45 [INFO] Device reset

chuyao, 2024-06-14:
Hi, I found that `Output::new(p.PB8, Level::Low, Default::default())` can't work on ch32v103r8t6, then I changed to `Level::High`, it worked, that's confusing. why? LED should work on low level official demo board

chuyao, 2024-06-14:
Sorry, maybe that's my cable's problem, both `Level::Low` and `Level::High` can work normally. Anyway, tks

nemasu, 2025-04-23:
Hello, just something to add:
I had to add the "unsafe-trust-wch-atomics" feature to qingke to get it to compile.

Dummyc0m, 2025-04-23:
> Hello, just something to add: I had to add the "unsafe-trust-wch-atomics" feature to qingke to get it to compile.

Did you read the warning text for the feature?

nemasu, 2025-04-23:
> > Hello, just something to add: I had to add the "unsafe-trust-wch-atomics" feature to qingke to get it to compile.
> 
> Did you read the warning text for the feature?

Yeah. I read through [issue 9](https://github.com/ch32-rs/qingke/issues/9) in the qingke repo too.
I'm not really sure what I'm going to do going forward yet.