ch32-rs/ch32-hal #25: How to check final binary size?

acedogblast, 2024-06-11:
Cargo built generates an ELF executable file as a output format which contains many additional data that will not be transferred to the CH32XXX chip. Is there a way for me to know the exact size of our firmware binary so I can keep track of how much flash we are using? 

So far I found that cargo bloat can give me the size of the .text section of the binary's ELF format. Is there a better way?

romainreignier, 2024-06-12:
You can use `cargo size`:

```
cargo install cargo-binutils
rustup component add llvm-tools
cargo size --release
```

andelf, 2024-06-12:
The traditional method is `objcopy -O binary`.
In Rust, many new tools are invented.

`wlink` and `wchisp` also print firmware info when flashing.

closed 2025-03-24 by acedogblast