ch32-rs/ch32-hal #29: Can't build any embassy example at all!

jsprog, 2024-06-18:
I'm having a hard time trying to make at least one embassy example work. Unfortunately, I confirmed that none of the official examples worked, event with the newer cargo-generate template (ch32-hal-template). According to other issues and discussions in this repository, it appears that this problem is unlikely to be affecting every developer! probably **proc-macro** incompatibilities across nightly revisions!

The error messages bellow were received after trying to build `examples/ch32v003/src/bin/embassy_blinky.rs` with `cargo build --release --bin embassy_blinky.rs` or `cargo build --bin embassy_blinky.rs` and I confirmed that I'm receiving the same error messages with every other embassy example regardless of the target device.

- nightly-x86_64-unknown-linux-gnu - rustc 1.81.0-nightly (59e2c01c2 2024-06-17)
- extracted the download from https://github.com/ch32-rs/ch32-hal (current-rev: 56de16c) as it is, and inspected the the examples from their current directories. the non embassy examples I tested worked perfectly.

**Build errors**
```console
error: item does not constrain `blink::Fut::{opaque#0}`, but has it in its signature
  --> src/bin/embassy_blinky.rs:11:1
   |
11 | #[embassy_executor::task(pool_size = 2)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: consider moving the opaque type's declaration and defining uses into a separate module
note: this opaque type is in the signature
  --> src/bin/embassy_blinky.rs:11:1
   |
11 | #[embassy_executor::task(pool_size = 2)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `embassy_executor::task` (in Nightly builds, run with -Z macro-backtrace for more info)
```

```console
error: item does not constrain `__embassy_main::Fut::{opaque#0}`, but has it in its signature
  --> src/bin/embassy_blinky.rs:23:1
   |
23 | #[embassy_executor::main(entry = "qingke_rt::entry")]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: consider moving the opaque type's declaration and defining uses into a separate module
note: this opaque type is in the signature
  --> src/bin/embassy_blinky.rs:23:1
   |
23 | #[embassy_executor::main(entry = "qingke_rt::entry")]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `::embassy_executor::task` (in Nightly builds, run with -Z macro-backtrace for more info)
```

**More Details from Rust Analyzer**
```console
proc-macro panicked: failed to load macro: Failed to get file metadata for /home/developer/dev/ch32-hal/examples/ch32v003/target/debug/deps/libembassy_executor_macros-ca73f4ef56da3f51.so: No such file or directory (os error 2)rust-analyzer[macro-error](https://rust-analyzer.github.io/manual.html#macro-error)
```

jsprog, 2024-06-18:
Already reported by others here https://github.com/embassy-rs/embassy/issues/3078 and likely the issues is affecting embassy since nightly-2024-06-13

andelf, 2024-06-19:
Thanks. A nightly rust-toolchain lock should be added before embassy's next crates.io release.

andelf, 2024-06-20:
Added `rust-toolchain.toml` for a temporary workaround.

sb-child, 2024-07-07:
Another temporary workaround is:

lure23, 2024-07-27:
A third way to work around:

```
$ cargo +nightly-2024-06-01 run --release
```

I prefer this, due to the problem likely being temporary in nature.

Codetector1374, 2024-10-01:
The problem here is that qingke_rt requires nightly because it uses naked_function. This can be worked around by generating some assembly wrapper like thr stm32 stuff does. But it is currently not implemented

andelf, 2024-10-01:
@Codetector1374 Thank you for the suggestion. I'm currently on holiday, but I plan to implement this when I return.

Of course, pull requests are always welcome. 😊

andelf, 2024-10-30:
Close as fixed.

closed 2024-10-30 by andelf