ch32-rs/ch32-hal #32: Limitations to PWM frequency?

radwouters, 2024-06-23:
I don't know if the following is an issue specific to the implementation in this HAL, or that it is a limitation in the chip, so please let me know if this issue is in the wrong place.

For a project in which I am sending [38 kHz NEC messages](https://techdocs.altium.com/display/FPGA/NEC+Infrared+Transmission+Protocol) on a CH32V003F4P6 dev board, I am using SimplePWM to generate a 38 kHz PWM signal and then modulate this by calling `pwm.enable` and `pwm.disable`, as follows:
```rust
    let mut pwm = SimplePwm::new(
        p.TIM1,
        None,
        None,
        None,
        Some(pin),
        Hertz::khz(38),
        CountingMode::default(),
    );
    let ch = hal::timer::Channel::Ch4;

    // Futher, in the loop to send bits:
    pwm.enable(ch);
    Delay.delay_us(563);
    pwm.disable(ch);
    if bit == 0 {
        Delay.delay_us(563);
    } else {
        Delay.delay_us(1688);
    }
```
When looking at an oscilloscope connected to the output pin, it looks as follows:
[image]
[image]

Is this a limitation of the CH32V003? That seems unlikely to me, given that the clock of a CH32V003 is about 1000 times higher than the signal frequency? Did I configure the PWM wrong?

I see that in the WS2811 example the signal is bitbanged in code as well, was that done for the same reason?

If any additional information is needed, please let me know :) And thanks for the great work on this project!

andelf, 2024-06-23:
The FLASH accessing speed limits the speed of CH32V family MCUs.
So `Delay.delay_us` might be unreliable.

`#[highcode]` is used in `gpio_ws2812.rs` demo, which moves code to SRAM. I'll add a timer-driven example soon.

radwouters, 2024-06-23:
I think that this behaviour can be expected and I just did not think about it properly: Only the first pulse is deviating in width
[image]
Which of course is pretty logical, since when the pwm channel gets enabled, the timer can be in the middle of a cycle, but the oscilloscope only gets triggered once the channel is enabled. Therefore, it sees only the tail end of that cycle.

If you have an alternative solution, I'd love to see that, but I think that for my application the odd first pulse won't be too much of an issue. 

Many thanks for your quick response :)

closed 2025-06-25 by Dummyc0m