ch32-rs/ch32-hal #39: [BUG] Trait bound not satisfied for usbd-hid-macros crate

jake-g00dwin, 2024-07-19:
Ran into an issues with building and testing the examples for the ch32v203. This is what I'm getting when I try to run `cargo build --bin blinky`.

```sh
error[E0277]: the trait bound `usbd_hid_descriptors::MainItemKind: From<std::string::String>` is not satisfied
   --> /home/ronin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/usbd-hid-macros-0.6.0/src/spec.rs:512:43
    |
512 |             self.set_item(name, item_kind.into(), settings, bits, quirks);
    |                                           ^^^^ the trait `From<std::string::String>` is not implemented for `usbd_hid_descriptors::MainItemKind`, which is required by `std::string::String: Into<_>`
    |
    = help: the trait `From<&str>` is implemented for `usbd_hid_descriptors::MainItemKind`
    = help: for that trait implementation, expected `&str`, found `std::string::String`
    = note: required for `std::string::String` to implement `Into<usbd_hid_descriptors::MainItemKind>`

   Compiling futures-util v0.3.30
For more information about this error, try `rustc --explain E0277`.
error: could not compile `usbd-hid-macros` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
```

It's coming from the GroupSpec Impl in the USBD HID macros crate.

jake-g00dwin, 2024-07-19:
Okay found the upstream issue: [usbd-hid issue 73](https://github.com/twitchyliquid64/usbd-hid/issues/73)
and [embassy issue 3078](https://github.com/embassy-rs/embassy/issues/3078)

Basically it's an issue with embassy and dependencies user older versions of the crate when the 0.8 version is the only one that really works right now.

Using the nightly-2024-06-12 tool chain seems to still run into the same issue as of right now.

jake-g00dwin, 2024-07-22:
Had some time to mess with it again today, managed to find a workaround. Using the examples for the ch32v203 and the nightly tool-chain from 2024-06-11 I managed to get it to build and compile.

I had to first edit the `Cargo.toml` file so that it used the other embassy line and also so it didn't include the usb dependency.

```toml
//---SNIP---
#embassy-usb = { version = "0.1.0" }
//---SNIP---
[patch.crates-io]
embassy-executor-macros = { git = 'https://github.com/embassy-rs/embassy'}
//---SNIP---
```

Final thing I noticed is that some examples required me to add a line for a feature dealing with implementing traits for associated types.

```rs
#![feature(impl_trait_in_assoc_type)]
```

After adding that line I can compile and flash the blinky, adc and the uart(serial) programs without any issues.

andelf, 2024-07-23:
I'm planing to change embassy-related deps to git version.

twitchyliquid64, 2024-07-24:
Worth noting theres a `0.6` and `0.7` patch release that fixes this if you don't want to update to `0.8` crates from `usbd-hid`.

jake-g00dwin, 2024-07-24:
@twitchyliquid64 Nice, that sounds like a great temporary solution as well.

closed 2025-06-25 by Dummyc0m