ch32-rs/ch32-hal #40: USB device example

kekcheburec, 2024-08-28:
Hi, @andelf !

I saw that in this hal there is a USB device implementation, but there is no example for working on the same ch32v203 :(

Or is the implementation not working at all?

andelf, 2024-08-28:
I haven't tested it yet. CH32V has two completely different USB IP cores. One is USBD used in CH32V203, which is almost the same as STM32's. So I just copied the implementation.

Another IP core is USBH(OTG) used in some other chips.

Dummyc0m, 2024-09-14:
Do you have any plans to work on the OTG peripheral in the near future? If not, we're interested to help out and avoid duplicate work.

kekcheburec, 2024-09-14:
Hi @Dummyc0m !

Personally, at the moment I do not plan to implement usb otg peripherals. I just wanted to find out if what is already in the repository works and got an answer.

In general, you can close the issue.

Dummyc0m, 2024-09-14:
Sorry, I was actually hi-jacking the issue and asking @andelf to see if they have any plans.

Vixea, 2024-09-14:
Why close the issue as complete? There's no USBD example