ch32-rs/ch32-hal #43: Support for BLE on the CH32V208

Frostie314159, 2024-09-03:
Hi, is BLE support for the CH32V208 planned?

andelf, 2024-09-03:
Thank you for your interest. 

BLE support for the CH32V208 is indeed planned. However, it's a complex task that requires some development effort, and its implementation may not be immediate. 

I'll update when I have more concrete plans on the timeline.

Currently, https://github.com/ch32-rs/ch58x-hal has a proof-of-concept implementation of BLE on CH582 MCU.
The BLE APIs are almost the same as CH32V208's.

Frostie314159, 2024-09-03:
Hi, thanks for the quick response! Could you potentially send me an ELF binary of one of the C example programs for BLE? I don't have a working build environment, and I'd like to examine the binary. Potentially, we could reverse engineer the registers and build a rust only BLE stack.

andelf, 2024-09-04:
The official BLE libs are at <https://github.com/openwch/ch32v20x/tree/main/EVT/EXAM/BLE/LIB>.

ReFil, 2026-10-07:
Id love to see this too, the v208 looks really neat for a basic usb + ble application