ch32-rs/ch32-hal #45: Is the interrupt::typelevel::Handler wrong?

Codetector1374, 2024-10-01:
Looks like the `interrupt::typelevel::Handler<T>` type actually has this `unsafe fn on_interrupt` interface which is directly linked into the interrupt vector table. But the function is not actually wrapped in `qingke_rt::interrupt` thus it use `ret` instead of `mret`.

Has anyone tried using anything beside EXTI that uses interrupt? It looks like it will just put the core into interrupt mode and never leave? (Including the Async UART Tx/Rx ?)

andelf, 2024-10-02:
Thanks. It should be marked as `#[qingke_rt::interrupt]`

andelf, 2024-10-02:
Thanks.
This is a significant typo that needs to be corrected. 🥲

Codetector1374, 2024-10-06:
Looks like this got fixed by the above commit. Closing

closed 2024-10-06 by Codetector1374