ch32-rs/ch32-hal #52: V203 with embassy: awaits infinitely

Dicklessgreat, 2024-10-28:
My current program:
```
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]
#![feature(impl_trait_in_assoc_type)]

use ch32_hal::println;
use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use hal::gpio::{Level, Output};
use {ch32_hal as hal, panic_halt as _};

#[embassy_executor::main(entry = "qingke_rt::entry")]
async fn main(_spawner: Spawner) -> ! {
    hal::debug::SDIPrint::enable();
    println!("hello!");
    let p = hal::init(Default::default());

    let mut led = Output::new(p.PA15, Level::Low, Default::default());
    loop {
        led.toggle();
        println!("toggled!");
        Timer::after(Duration::from_secs(2)).await;
    }
}

```
Cargo.toml:
```
[dependencies]
panic-halt = "1.0.0"
qingke-rt = "0.4.0"
qingke = "0.4.0"
ch32-hal = { git = "https://github.com/ch32-rs/ch32-hal", features = [
    "ch32v203c8t6",
    "defmt",
    "embassy",
    "rt",
    "time-driver-tim2",
], default-features = false }
embassy-executor = { version = "0.6.0", features = [
    "integrated-timers",
    "arch-riscv32",
    "executor-thread",
] }
embassy-time = { version = "0.3.2" }
embassy-futures = { version = "0.1.0" }

```
output from `wlink`:
```
2024-10-29 03:16:42.470: hello!
2024-10-29 03:16:42.471: toggled!
```
and no more outputs.(I expected bunches of "toggled!")
Connected led does not blink.
And I also tried `ch32-hal-template`, but it also doesn't blink.
And I tried ch32v203's blink example, which does not use `embassy`, but use `qingke` crate only, then, it blinks correctly.

I think something's wrong with ch32v203's embassy features?

jake-g00dwin, 2024-10-29:
I've just tested and ran the existing SDI example from the repo and it works fine. I'm testing this on a ch32v203k8t6 micro-controller.

I think you're missing something pretty basic in your example code. You're trying to use Embassy stuff without actually using it.

You need to initialize it. This isn't an issue with this repo. I would really suggest reading the documentation for embassy.

**File:** sdi_print2.rs
```rs
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]
#![feature(impl_trait_in_assoc_type)]

use ch32_hal::println;
use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use hal::gpio::{Level, Output};
use {ch32_hal as hal, panic_halt as _};

#[embassy_executor::main(entry = "qingke_rt::entry")]
async fn main(_spawner: Spawner) -> ! {
    let p = hal::init(Default::default());

    //Think you forgot this part.
    hal::embassy::init(); // <---- THIS RIGHT HERE!
    // Look above.

    println!("Flash size: {}kb", hal::signature::flash_size_kb());
    println!("Chip UID: {:x?}", hal::signature::unique_id());
    let chip_id = hal::signature::chip_id();
    println!("Chip {}, DevID: 0x{:x}", chip_id.name(), chip_id.dev_id());

    hal::debug::SDIPrint::enable();
    println!("hello!");

    let mut led = Output::new(p.PB0, Level::Low, Default::default());
    loop {
        led.toggle();
        println!("toggled!");
        Timer::after_millis(5000).await;
    }
}
```

commands used to run:

```sh
cargo build --release --bin sdi_print2
wlink flash -chip ch32v20x ./target/riscv32imac-unknown-none-elf/release/sdi_print2
wlink sdi-print enable --chip ch32v20x
```

Dicklessgreat, 2024-10-29:
Thanks!
With `hal::embassy::init();`, it works fine!

closed 2024-10-29 by Dicklessgreat

andelf, 2024-10-29:
Thanks the the report. I'm planning to add `embassy::init` to `hal::init`, behind the `embassy` feature gate.