ch32-rs/ch32-hal #68: Can't degrade_adc  for PD3/A4 (ch32v003)

jsprog, 2024-11-09:
#### Description
AdcChannel trait isn't implemented for PD3/A4
```rust
    let adc_channel_7 = p.PD4.degrade_adc(); // success
    let adc_channel_4 = p.PD3.degrade_adc(); // not available
```
```
error[E0599]: no method named `degrade_adc` found for struct `PD3` in the current scope
```
#### Affected devices:
- `ch32v003f4p6` and `CH32V003f4u6`

jsprog, 2025-05-10:
@andelf 
Is this dead?

romainreignier, 2025-05-12:
Hi @jsprog 
Thanks for the report, there was an error in the peripheral definition, I have fixed it here: https://github.com/ch32-rs/ch32-data/pull/17

In the meantime, you can fix it locally in ch32-data, generate a new ch32-metapac with:
```
./d gen
```

And then use your local ch32-metapac in ch32-hal and then use this local ch32-hal in your project.

polygon, 2025-06-25:
Thank you, I just stumbled over this and the issue was fixed swiftly. Bummer that the project seems on kind of a hiatus right now.

closed 2025-06-25 by Dummyc0m