ch32-rs/ch32-hal #98: Bug ( Fixed ): program runs incorrectly when `pool_size` is greater than `1`

Yttehs-HDX, 2025-05-18:
## My Env

| MCU Chip | Rust Version | Dev Env |
|:---:|:---:|:---:|
| VT-CH32V307VCT6 | Nightly ( newest ) | ArchLinux x86_64 |

## Description

- At cocurrent task defination:

```rust
#[embassy_executor::task(pool_size = 2)]
async fn name(...) {
    // ignore
}
```

All tasks failed to run, including loop logic in main function ( `#[embassy_executor::main(entry = "qingke_rt::entry")]` ).

## Way to Fix

1. At `Cargo.toml` from current project.
2. Find `dependencies.embassy-executor`.
3. Edit:

```diff
embassy-executor = { version = "0.6.1", features = [
    "integrated-timers",
    "arch-spin",
    "executor-thread",
-    "task-arena-size-192", # or better use nightly, but fails on recent Rust versions
+    # "task-arena-size-192", # or better use nightly, but fails on recent Rust versions
+    "nightly",
] }
```

Then cocurrent tasks run correctly.

## Help Wanted

Is there better ways to fix the `cocurrent tasks` issue? Thanks.

romainreignier, 2025-05-18:
Hi @Yttehs-HDX 
Thanks for the report.
The minimum size of 192 was picked in the [template](https://github.com/ch32-rs/ch32-hal-template) to work for the small CH32V003 parts.
When adding more tasks, you need more stack and can increase the size for the list [here](https://docs.rs/embassy-executor/latest/embassy_executor/#task-arena-size). The panic should display it if your probe is able to show it to you.
But as you have found out, when using a nightly toolchain, we can use the `embassy-executor` `nightly` feature.
Since the removal of atomics in https://github.com/ch32-rs/ch32-hal/pull/61 we are forced to use a nightly toolchain anyway to define a custom target so your suggestion is valid.
For info, the next release of `embassy-executor` will be able to compute the exact stack size on stable toolchain thanks to this change https://github.com/embassy-rs/embassy/pull/4020.

Yttehs-HDX, 2025-05-19:
Thanks for the answer @romainreignier 

## To Summarize

You can define the stack size by choosing another feature from [embassy-executor](https://docs.rs/embassy-executor/latest/embassy_executor/#task-arena-size), at _Preconfigured Task Arena Sizes:_,

or simply use `nightly` feature anyway.

And the most important is that, be careful of the **Stack Overflow**.

closed 2025-05-19 by Yttehs-HDX